pulseq/pulseq #1: git path correction?

pefs44, 2016-05-07:
Hi, 

   glad to see you building the dream of pulse sequence development outside the proprietary environments. 

    For me the download instruction  on: http://pulseq.github.io/index.html
git clone git@github.com:pulseq/pulseq.git   fails with messages: 

RSA key fingerprint is 16:27:ac:a5:76:28:2d:36:63:1b:56:4d:eb:df:a6:48.
Are you sure you want to continue connecting (yes/no)? y
Please type 'yes' or 'no': yes
Warning: Permanently added 'github.com,192.30.252.129' (RSA) to the list of known hosts.
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights and the repository exists.
(I don't think it should matter, but I'm on a OSx 10.95 machine). 

I had more success with: 
git clone https://github.com/pulseq/pulseq.git

Paul Summers

kelvinlayton, 2016-05-09:
Hi Paul,

Thanks for your interest in our project! To use the path git@github.com you need to setup SSH keys in your github account (see https://help.github.com/articles/generating-an-ssh-key/). This is intended for developers who wish to contribute to the project. To simply use the code, you can use the https protocol, as you did, or download the ZIP file directly.

Note the standard method to contribute code is to fork the project into your own account and then generate pull requests. Otherwise, please let us know of any bugs or feature requests!

Hopefully that helps.
Kelvin Layton

pefs44, 2016-05-09:
thanks Kelvin, 
until I get some of the rust out of my coding skills, I'll be on the user side.
Paul

closed 2016-05-09 by pefs44