pulseq/pulseq #4: Raster time check on mr.makeDelay would help

sairamgeethanath, 2018-10-16:
A check on whether the prescribed delay in the delay block is a positive integer multiple of gradient  rastertime would help in catching errors where delay might be computed elsewhere or in other parts of the code. Alternately ceiling/rounding with the developer's knowledge may also be considered.

stefan-k, 2018-11-30:
Thanks. We have also noticed this problem. Unfortunately the system limits need to be propagated in order to do this. Many functions already allow the limits to be passed in, but this is not very ergonomic. We are thinking about a different design where all event-creating functions are methods of the Sequence class. That way all system limitations are readily available.

m-a-x-i-m-z, 2021-05-14:
seq.checkTiming() does this kind of checks in the current public version

closed 2022-09-06 by m-a-x-i-m-z