pulseq/pulseq #13: Max. linesize during Sequence parsing from external file too small

lukas-k, 2021-07-09:
Dear all,

sometimes we are using attributes in the [DEFINITIONS] section which are longer than the maximum line size of 256 as defined in ExternalSequence.cpp as:

`const int ExternalSequence::MAX_LINE_SIZE = 256;`

Hence, definition attributes are read incorrectly when using the function GetDefinition(attribute).

I think, extending this value (e.g. to 512) is a rather uncritical step not interfering with any storage limitations or specifications. Of course we can change it locally, but this would cause extra maintenance effort and might be helpful for other users, too, who are wondering about truncated lines.

If you think this little request is rather insignificant, please close the issue without any changes.

m-a-x-i-m-z, 2021-07-14:
Hi Lukas, 

I accept that 256 characters is quite restrictive, but on the other hand any higher constant may be seen by somebody as to restrictive at some point. A better solution would be one of the two: 

* either change the code to adapt to the line length (can be done in multiple ways);
* or extend the format specification to allow for line breaks in definitions e.g. by adding a backslash character at the end of the line. 

I have to think about it a bit to see how to handle this situation optimally 

Best regards, 
M.Z.

m-a-x-i-m-z, 2021-07-14:
OK, I was able to implement a relatively simple fix for the definitions of arbitrary length, next version (1.4.0) will have a proper fix. 
--Maxim

m-a-x-i-m-z, 2022-09-06:
In v1.4.0 and later the issue is fixed.

closed 2022-09-06 by m-a-x-i-m-z