pulseq/pulseq #16: Loading a Pulseq file containing 400,000 TRs

namgyunlee, 2022-08-21:
I am trying to run a free-breathing pulse sequence with Pulseq.

My pulseq file contains about 400,000 TRs based on spiral phyllotaxis radial trajectories (ECG-gated, dual-echo, radial bSSFP pulse sequence with a TR of 2.4 ms).

It seems I can't load a pulseq file containg more than 16,000 TRs at a ramped-down 0.55T Siemens Aera system.

One possible solution I can think of is to split 400,000 TRs into multiple pulseq files but this may not be an optimal solution because when a new pulseq file is played, magnetization needs to be driven into steady-state again.

Do you have any alternative solution? 

I appreciate any help. Thank you so much!

m-a-x-i-m-z, 2022-08-24:
Hi, 

there seem to be an issue with the memory allocation in the libBalance library of Siemens. It may help to disable it either on the Sequence/Special card or even completely at the compile time.  Please let me know if that helps. 

Best regards, 
Maxim

bilal-tasdelen, 2022-08-27:
@m-a-x-i-m-z Disabling libBalance from the Sequence card improved the preparation times, but did not alleviate the issue.

Compiling the sequence without libBalance support solved the incremental slow-down, and loading large sequences does not crash the scanner anymore. 

Thanks for the help and your time.

namgyunlee, 2022-08-28:
@m-a-x-i-m-z Bilal and I were able to run a Pulseq file containing 160K TRs very smoothly, without any hustle. Thank you so much for the help ;)