pulseq/pulseq #17: Multiple block pulses in a single event block

bilal-tasdelen, 2022-10-13:
It looks like adding multiple block pulses in a single event block only adds the latest pulse and discards the rest. Is this the expected behavior?

Adding multiple RF pulses at once is really convenient for creating pulse trains and gapped pulses.

Here is a minimally reproducible sequence and associated diagram:

```matlab
addpath('../pulseq/')

% =============================================================================
% SETUP
% =============================================================================

% Set system limits

sys=mr.opts('maxGrad', 30, 'gradUnit', 'mT/m', ...
            'maxSlew',140, 'slewUnit', 'T/m/s', ...
            'gradRasterTime', 10e-6, ...
            'rfRingdownTime', 10e-6, ...
            'rfDeadTime', 10e-6, ...
            'rfRasterTime', 1e-6, ...
            'adcDeadTime', 10e-6);

% =============================================================================
% DERIVED SEQ PARAMS
% =============================================================================

dw = 80e-6;
trf = 30e-6;
Trf = 4e-3;
Nseg = ceil(Trf/dw/4)*4;
Trf = Nseg*dw; % Actual Pulse duration

% =============================================================================
% CREATE EVENTS
% =============================================================================
seq = mr.Sequence(sys);           % Create a new sequence object

rf_list = {};

for rf_i = 1:Nseg
    rf_list{rf_i} = mr.makeBlockPulse(pi/8, ...
        'duration', trf, ...
        'delay', dw*rf_i, ...
        'use','excitation', ...
        'system', sys);
end

% =============================================================================
% CONSTRUCT SEQUENCE
% =============================================================================

for i = 1:10
    seq.addBlock(rf_list);
end

[ok, error_report] = seq.checkTiming;

if ok
    disp("Timing check passed successfully")
else
    disp("Timing check failed! Error listing follows:")
    disp(error_report)
end

seq.plot('TimeRange', [0, 1.1], 'TimeDisp', "ms")
```

[image]

m-a-x-i-m-z, 2022-10-21:
Hi! 

adding multiple RF pulses to a single block is prohibited by design, please see https://pulseq.github.io/specification.pdf 
There are two options to work this around: 
* splitting the sequence into several blocks (the inter-pulse spacing will then be limited by the sum of the dead and lead RF times) 
* combining RF pulses into one long RF pulse (which may be limited by the total number of samples in the RF pulse)
If you follow the second route you may offset the limitations somewhat by using a larger RF dwell time for that pulse. 

The fact the Pulseq accepts multiple RF pulses in the same block quietly is a bug. It mus throw an error. 

Best regards, 
Maxim

bilal-tasdelen, 2022-10-21:
Thanks for the explanation! 1st option seems to be a better solution for my case. Closing.

closed 2022-10-21 by bilal-tasdelen