pulseq/pulseq #19: [Siemens interpreter sequence] RF Amplitude not exact error when patient table move requested.

bilal-tasdelen, 2022-10-29:
Some .seq files with high RF amplitude causes sequence to be invalid in following situation:

- Timing and Flip Angles = strict
- Patient Table Mode = ISO or FIX
- A patient table shift more than 15 mm is requested.

Settings:

[image]

This is fine:

[image]

Making it 16 mm is not fine:

[image]

Error from logviewer:

[image]

This happens because, when a patient table movement of 15mm or more is requested, Tx calibration is invalidated. This causes reference Tx amplitude to be 0,  and sets  the actual RF amplitude to a default 170V. Thus, isFlipAngleExact() function returns false, and sequence becomes invalid.

This is fine (with 15mm shift):

[image]

This is not fine (with 16 mm shift):

[image]

Thanks!

m-a-x-i-m-z, 2023-10-20:
Hi!

This is an interesting issue and I was wondering, how product sequences handle such situations. Could somebody do some more testing? At some point after the sequence is started and the table has been moved the scanner should have re-calibrated its transmitter's reference voltage and the sequence should be prepared again. How could we recognize this moment from the sequence's perspective? 

--Maxim

btasdelen, 2023-11-17:
Hi @m-a-x-i-m-z,

Do you have a specific test do you want me to perform? 

I tested this with several product sequences, I am pretty sure it is independent of the sequence and independent of the ISO mode as well, as soon as the table moves more than 15mm, a transmit calibration is triggered. I don't see the `isFlipAngleExact()` function in any of the sequences we have that was shared by Siemens, so it looks like they don't handle it at all, they just let it recalibrate.

If we are able to get the patient table position inside the sequence, can we simply track it and bypass the `isFlipAngleExact()` function if there is a movement more than 15mm?

m-a-x-i-m-z, 2023-11-18:
Hi Bilal, 

thanks for the detailed info, I'll try to look into this in the coming weeks. Maybe in the mean time you can help me out by trying to find out how many times the prep() function is called after the table was moved. 

The thing is that there is actually no guarantee that the flip angle remains reachable after the recalibration because the table movement may potentually change the coil loading and if the reference voltage goes up, some of the pulses may get trimmed... Strictly speaking, we would like to be able to detect this situations, don't we? 

Best regards, 
Maxim