pulseq/pulseq #20: Pulseq compile error on VA50

jsharmaha, 2022-11-21:
Hi,
I am getting errors while compiling pulseq for VA50. attached is the error.
Any fixes?
Thanks,
Harish
[image]

m-a-x-i-m-z, 2023-03-27:
Hi, 

we have a patch to 1.4.0. in the lab but we are also working on preparing a new release (1.4.1) which should resolve it. 

Best regards, 
Maxim

cncastillo, 2023-05-30:
I know this issue is closed, but I think this information could be useful. 

I just compiled the newest Pulseq 1.4.1 for the XA50. I was also getting the error presented above, so to make it work I had to add `CPPFLAGS (-DDUMMY_MPCU_TRACE_IN_USE)` to the makefile (so it didn't try to import `nlsmac.h`) and also had to create a "../parameter_map/" folder containing "parameter_map3.h". 

On the other hand, to simulate the sequence I had to change the "FOV positioning" to "disabled" or any other option that didn't include translation of the FOV. I assume this doesn't work for the simulator, but it does for a real scanner (I will try soon). After that, everything is working as expected.

Is this the recommended way of doing it? If so, maybe it could be further explained in the C2P Manual. Any insight on why the FOV positioning could be generating a problem?

Cheers!

m-a-x-i-m-z, 2023-06-06:
Interesting... 
Looks like I've missed to commit a set of small changes in this small changes in yo our local git server prior to the release. I'll double check and will post an update here. Reopening the issue for the time being... 
--Maxim

m-a-x-i-m-z, 2023-07-28:
This is issue should be fixed in the current interpreter.

closed 2023-07-28 by m-a-x-i-m-z