pulseq/pulseq #23: RF Raster time change from 1us to 2us

namgyunlee, 2023-04-06:
Dear Maxim,

I have just simply changed a RF raster time from 1us to 2 us in Siemens.

However, an RF raster time of 2us gave a noiselike image while an RF raster time of 1 us gives a nice image as expected.

Do you suggest any clues why changing RF raster time does not work?

Thank you.

Sincerely,
Nam Lee

Attached is a screenshot comparing 2us vs 1us images.
[image]

m-a-x-i-m-z, 2023-04-06:
Dear Nam Lee, 

you are not supposed to change these things this way, actually we may take this as a bug report to make the interpreter to enforce the match between the declared RF raster and the one on the system. 

I wonder whether you are aware of the correct way of doing what you seem to be trying to achieve. To define RF shapes with longer durations and smaller numbers of support points you should use the 'dwell' parameter of the makeArbitraryRF() or makeSincPulse(). 

Best regards, 
Maxim

namgyunlee, 2023-04-06:
Dear Maxim,

Thank you for your explanation.

This attempt (changing from 1us to 2us) was based on my previous success on BIR-4 implementation.
BIR-4 pulses with a longer duration were only realized in a scanner when an RF raster time was switched from 1us to 2us (RF excitation was a hard pulse so I was not able to catch any abnormalities).
For a short duration (5 or 6 ms), BIR-4 was possible without this trick.

I will try a BIR-4 pulse with an RF raster time of 1us and see if I still have an error.

Thank you!

Sincerely,
Nam

m-a-x-i-m-z, 2023-04-06:
Hi Nam, 

I'm not sure I understand the problem. On one hand the Siemens interpreter sequence should automatically down-sample your pulses if they get longer than 8192 us with the default dwell time of 1 us. So, basically your sequence should run with the default raster without any noticeable problem. However, you may even avoid this by explicitly specifying the dwell time (or the number of samples), see for example the semiLaser sequence (https://github.com/pulseq/pulseq/blob/master/matlab/demoSeq/writeSemiLaser.m) 

Hope this helps!
Maxim

namgyunlee, 2023-04-09:
Dear Maxim,

Thank you for providing the useful information. It was a great help. :)

"On one hand the Siemens interpreter sequence should automatically down-sample your pulses if they get longer than 8192 us with the default dwell time of 1 us"
=> I was totally unaware of this fact. I think this information is very important. 
 
I will take into account this fact and redesign my BIR-4/BIR-8 implementations.

Sincerely,
Nam

closed 2023-04-09 by namgyunlee