pulseq/pulseq #24: Seq with large number of slices loaded to Siemens

Bochao0331, 2023-04-29:
Hi Pulseq fellows! I tried to load a seq of,  **60 slices** (for the 60 spectral bins in my EPSI: 60 x 0.2 kHz = 12kHz) with 4 readouts, slicethickness = 1.5mm, to the Siemens scanner, but it showed the following error. If I change **to 1 or 20 slices** for test, the seq file can be loaded. Could it be relevant to the maximum slice number set in ExternalSequence.cpp (similar to the discussion in https://github.com/pulseq/pulseq/issues/13). If so, where can I find the ExternalSequence.cpp, and should we change it?

[image]

Bochao0331, 2023-04-30:
Thank you, Maxim! Good to know this library. Maybe I can provide more information: the _slice_ loop is inside the _phase encoding_ loop for each TR, so for either 1, 20 or 60 slices,  the running time is the same (nPE * TR). Do you think it's still related to longer running time?

Bochao0331, 2023-05-01:
Hello @m-a-x-i-m-z , the seq file was successfully loaded for no particular reason. However, it seems that using this code creates multiple slice groups from the scanner UI, rather than multiple slices within one slice group (`lbl_set_slc = mr.makeLabel('SET', 'SLC', slc - 1)` used in my slice loop). This can be quite tedious when trying to change the slice locations or FOV, cuz I have to open each slice group and change one-by-one. Do you think this is a bug?

m-a-x-i-m-z, 2023-05-04:
Hi!

great to hear that the sequence works now! BTW, do you have an estimate of the critical sequence duration (or the number of blocks) when libBalance starts to crash? 

As for the slice locations: this is not a bug but is in fact an intended behavior. This was the only way of getting multi-slice Pulseq sequences to  work together with the Siemens ICE image reconstruction. We are thinking on making it possible in future to reconstruct proper slice packages with correct slice positions, but this would require deeper changes in the code and potentially quite some debugging. Yo may consider starting another issue here on GitHub as a feature request dedicated to that. 

--Maxim

Bochao0331, 2023-05-04:
Hi Maxim,

Thanks for your reply! My labmate already disabled the libBalance (I'm from Krishna's lab as @namgyunlee and @bilal-tasdelen), so no running time issues are happening now. My running time is only 16 mins.

We found if we turn on 'Set for SLI', then multiple slice groups will change to 1 slice group and we can just change the slice location once.

m-a-x-i-m-z, 2023-05-04:
Hi, 

yes, that's true, we do have the "SET for SLC" feature, I didn't think of it for a moment. Good point! Happy to hear that you've found a solution!

--Maxim

Bochao0331, 2023-05-04:
Thank you for the discussion! I will close this !

closed 2023-05-04 by Bochao0331