pulseq/pulseq #25: Questions about RF part of `waveforms_and_times`

bilal-tasdelen, 2023-05-05:
Hi,

We are having a problem with `waveforms_and_times` RF implementation at PyPulseq, so I wanted to ask some implementation choices to better understand the issue.

The whole discussion can be seen here: [https://github.com/imr-framework/pypulseq/pull/114](https://github.com/imr-framework/pypulseq/pull/114)

In summary, for long sequences, the following check fails and issues a warning:

From `Sequence.m`, `waveforms_and_times()`, line 1472:

```Matlab
if wave_cnt(j)~=length(unique(wave_data{j}(1,1:wave_cnt(j))))
    warning('Warning: not all elements of the generated time vector are unique!\n');
end
```
This happens due to these lines:

From `Sequence.m`, `waveforms_and_times()`, line 1421-1426:

```Matlab
if abs(rf.signal(1))>0
    pre=[curr_dur+rf.delay+rf.t(1)-eps;0];
end
if abs(rf.signal(end))>0
    post=[curr_dur+rf.delay+rf.t(end)+eps;0];
end
```

The issue is, we are adding `eps` to the first time and last time point, but `eps` is only correct when floating point is 1.0. For time vector, we can have large values, so the difference between two floats is larger than `eps`. This causes the added `pre` and `post` time points to be the same, rather than `eps` apart.

The following explanation is from numpy's documentation:

> eps: float
> 
> The difference between 1.0 and the next smallest representable float larger than 1.0. For example, for 64-bit binary floats in the IEEE-754 standard, eps = 2**-52, approximately 2.22e-16.

**Disclaimer:** I actually did not try if the issue happens with Matlab Pulseq, but I expect it to do so, since the floating representations are the same.

My two questions are:

1. What is the reason for adding pre and post zeros to the RF, can't it start from non-zero values?
2. What is the reason for checking the whole time vector here with `unique(wave_data{j}(1,1:wave_cnt(j)))`.? `unique` is relatively slow, especially for large sequences. Wouldn't it be sufficient to check only the first and last samples we added? Would it make sense to move this check to the RF object creation, so that we only do the check once for each object?

Thank you for your time!

m-a-x-i-m-z, 2023-05-05:
Hi Bilal, 

first of all, you are right, the implementation of the RF export in waveforms_and_times() is a bit of a hack, because I have only used it sporadically for sequence visualization purposes. (I actually wonder in which way are you using it then). 

The reason why we need that additional samples in the RF shapes is that many of the RF pulses do not explicitly start and end with 0. Examples are Gaussian fat saturation pulse or a block pulse. If we don't add zeros the way we do the last point of the preceding RF pulse will get connected to the first point of the following RF pulse, producing very ugly plots and not representing at all what the hardware actually does. 

The trick with the eps is in fact a bit of a hack, and yes, you are right, it certainly becomes a source of a challenge in long sequences. To find a proper solution we may need to consider in a bit more detail what is actually happening with the shapes an objects on a very granular scale. To make sure that we are on the same page I paste a section from the Pulseq format documentation here: 

> For understanding the time raster alignment it is important to differentiate
> between the edges and centers of the time raster steps. All blocks and events
> are aligned to the corresponding edges of the time raster. Block duration
> is required to be multiple of BlockDurationRaster, therefore each block can
> only start and end at the edges of the block duration raster. Gradient events
> may only start at time points which are multiple of GradientRasterTime; for
> trapezoid events, ramp times and plateau duration may also only be equal to
> multiples of GradientRasterTime. In contrast to that, any default sampling
> points (e.g. sampling points of arbitrary-shape gradients) are aligned to the
> centers of the corresponding raster steps. The same convention applies to the
> default RF shape sampling and ADC sampling points. This means that the
> time of the n-th point on the RF pulse shape, gradient shape or ADC sampling
> point can be calculated as t n = t start + ∆t(0.5 + n).

Now we have to find a way of handling two distinct cases: (a) normal shaped pulses, e.g. Gaussian, and (b) block pulses. The later for performance reasons do not store the shape but contain two RF samples of the equal amplitude corresponding to the two time points -- beginning and end of the block pulse. The discrete time points in the case (a) are on the time raster shifted by 1/2 of the RF raster time (centers of the time raster steps), whereas the time points in the case (b) are on the edges of the time raster steps. So, strictly speaking the two cases need to be handled separately. You are welcome to make a proposal :-) 

BTW, I'll commit a minor change to the dev branch of the Matlab code to avoid too often calls of unique() when I have time.

--Maxim

bilal-tasdelen, 2023-05-05:
Thanks for the detailed response!

Apart from the internal usage of the function in `calculateKSpacePP`, I have two other use cases. First one is Bloch simulation, combined with the `pts2waveform` function. The other one is visualization, I am trying to implement a plotting function that will be able to plot the whole sequence at once. Eventually, if we add a time range input, `plot()` function in sequence.m can also use `waveforms_and_times`.

The only trick I could think of is instead of `eps`, shifting by a much larger number, like `0.1*rfRasterTime`. Of course, your proposed solution seems much more proper.

For calling of unique, for now, I replaced it with `diff(wave_data{j}(1,1:wave_cnt(j))))` followed by a comparison with `eps`:

```Python
rftdiff = np.diff(wave_data[j][0, : wave_cnt[j]])
if np.any(rftdiff < eps):
    raise Warning(
                            "Time vector elements are not monotonically increasing."
                        )
```

This was 3 times faster than unique in Python. I hope it covers the edge  cases you had in mind.

Thanks!

m-a-x-i-m-z, 2025-02-26:
Actually, I'm thinking of switching to switching to discrete timing with a fixed precision for all internal calculations. My proposal would be to use a single 64-bin integer and store time in picoseconds. I figure this will be accurate enough for any NMR/MRI needs for the years to come and still will not place any major limitations on the total duration of the sequence, which would then be 9,223,372 seconds, which is a bit over 100 days.