pulseq/pulseq #29: 64 x 64 MRSI with 1024 readout samples does not work in 1.4.1

namgyunlee, 2023-07-28:
Dear Maxim,

My MRSI sequence of 64 x 64 phase encodes with 1024 readout samples does not work after upgrading Pulseq to 1.4.1. 

This sequence was fine at 1.4.0. 

I found out that the Siemens Pulseq interpreter calculates the total scan time correctly when loading a Pulseq file with 32 x 32 PEs:
32 x 32 x 2 averages x TR=0.5sec ~=17 mins.

However, the total scan time becomes ">24h" when a Pulseq file contains more than 32 x 32 PEs.

I appreciate any help.

Thank you so much.

Sincerely,
Nam Lee

m-a-x-i-m-z, 2023-07-28:
Hi Namgyun, 

would you be able to share a working and a not working Pulseq file with us? You may send them directly per e-mail to pulseq.mr -at- uniklinik-freiburg.de. 

Best regards,
Maxim

namgyunlee, 2023-07-28:
Dear Maxim,

Thank you so much for looking at this issue.

Here is the Pulseq file that worked for 1.4.0 but did not work for 1.4.1.
"Zhao_v2_100kHz_RT140us_TT520us_delay500us_amp6_Smax50_slice5.00mm_64x64.seq"
https://github.com/namgyunlee/replication_bstar/tree/main/seq_Zhao_2020_MRM

I have reduced the number of phase encodes from 64 x 64 to 32 x 32 to make it work.
"Zhao_v2_100kHz_RT140us_TT900us_delay500us_amp6_Smax50_slice5.00mm_32x32.seq"

I have used [this script](https://github.com/namgyunlee/replication_bstar/blob/main/seq_Zhao_2020_MRM/demo_pulseq_Zhao_2020_MRM.m) to create the Pulseq files.

Thank you so much.

Sincerely,
Nam Lee

m-a-x-i-m-z, 2023-07-31:
Hi, 

yes, we were able to confirm the bug, which was introduced in 1.4.1 leading to the internal overall scan time duration counter to overflow at LONG_MAX microseconds (approx. 40 minutes). Shorter sequences are not affected, which is why we were unable to detect the problem ourselves. We are testing the patch so see if all problems are solved and will update this thread once we see more data on this issue and its resolution. 

Best, 
Maxim

namgyunlee, 2023-08-01:
Dear Maxim,

The bug has been successfully resolved with a hotfix. We were able to run an MRSI sequence with 64 x 64 phase encodes and 1024 readout samples.

Thank you so much for your time and help.

Sincerely,
Nam Lee

closed 2023-08-01 by namgyunlee