pulseq/pulseq #32: ADC samples - when are they exactly sampled

mzaiss, 2023-09-04:
when plotting kspace trajectories  in pulseq, the adc samples indicate the exact position that is measured in the kspace, or when acquisition starts?

when we scan at a siemens system it seems the signal os actually axquired between two samples

m-a-x-i-m-z, 2023-09-04:
To the best of our knowledge Matlab toolbox plots theoretical sampled positions (assuming a perfect system calibration). This was confirmed ~two years ago in tedious experiments involving various ADC bandwidths, etc. You may consider consulting the section 2.6. of the format specification, which basically resulted from that research. Of course, if there are imperfectly calibrated delays, then the k-space position observed in the experiment may be different. 
--M.

mzaiss, 2023-09-06:
"This means that the
time of the n-th point on the RF pulse shape, gradient shape or ADC sampling
point can be calculated as tn = tstart + Δt(0.5 + n). "

this is hard to interpret, if i take this part of the spec then the crosses indicating the ADC samples are not at the mentioned tn
or am i missing something?

is deltat here the time between the adc samples, or the raster time?

QingpingChen1201, 2023-09-06:
Hallo,
Many thanks for your message. To the best of my understanding, there is a "hard of ADC dwell time" delay in Siemens due to hardware limits (RF coil warm-up, etc.). Therefore, the "real" sampled point lags the "theoretical" sampled point by Δt*0.5. Let's consider this situation - a sin(t) gradient waveform "theoretically" starts at sin(0) = 0. Then, the "actual" start point is sin(Δt*0.5) != 0,
Hope this makes it clearer a bit.
Best regards,
Qingping

m-a-x-i-m-z, 2023-09-06:
Hi Moritz, 

I would partly disagree with Qingping, we are not talking about hardware imperfection here but quite contrary, exact specification /convention. So, there is no difference between the real and the theoretical sampled point. Also for the example with the sinusoidal gradient -- the point with the amplitude of 0, e.g. sin(t=0), is actually never sampled. This is really important and is one of the intended messages of the corresponding section in the documentation.

Now, I've just checked with the FID sequence, the first sampled point shown by the Matlab's plot function has exactly the following time: 

t_adc1=t_blockStart+t_adcDelay+t_adcDwell/2

This is intended behavior and corresponds exactly to the intended message of the specification, I really don't see what is hard to understand there if one readily appreciates the concept of the raster/dwell time edges and centers. To that end I do agree that it would be helpful to create an illustration/figure. And, of course we are happy to change the formulation once we understand, what was unclear.  

Best regards, 
Maxim

mzaiss, 2023-09-06:
Hi Maxim,

i will try to read with more time and try to understand better.

For our experiment:
What would you expect if we run a single GRE line with gradient mom from
lets say -15 to 16

and then again inversed from 16 to -15

adc starts with flattop in both cases.

do you expect the two echoes to match, or to be shifted against each other?
Prisma

best,
Moritz

m-a-x-i-m-z, 2023-09-06:
yes, the samples will be shifted if I understand your example -- you basically do not sample the k-space center. But actually, if you want to do gradient echo with bipolar readouts, it's a good idea not to sample the k-space center but with a symmetric moment. You may check EPI with ramp sampling from the demo set or the tutorial from GRE to EPI to see how we do it... But I agree, this is an extremely tricky topic. 
--Maxim

mzaiss, 2023-09-07:
ok, part of our confusion can be adressed to pypulseq, so wrong forum. closed until we have read more ;)

closed 2023-09-07 by mzaiss