pulseq/pulseq #33: Image orientation and FOV shift inverted on our scanner. (How) can a fix be included into the interpreter code?

dabosch, 2023-09-07:
Hi,

I am using an exotic Siemens scanner (9.4T whole-body, software VE12U-AP04_F9)
With the standard pulseq interpreter (v1.4.1), all dimensions are inverted.
I added changes to the interpreter locally and am now wondering if there's a way to get them into upstream code without breaking other peoples' workflows.

I don't know if others with exotic scanners are reading here or even using pulseq. (CMRR? Neurospin? Maastricht?) Are you affected by this?

I'd be happy to get some comments and opinions on this. Is there any standard-way to include model-specific code into the interpreter?

Best,
Dario

This was needed to fix the image orientation:
in `ARBITRARY_SBB::SetOrientationMapping`:
change the sign of all (non-zero) entries of `rmta`

For correcting the FOV shift:
in `ARBITRARY_SBB::SetFreqCoefs`:
```
for (int i=0;i<3;i++){
    sliceOffCenter_siemens.v[i] *= -1;
}
```

m-a-x-i-m-z, 2023-09-07:
Hi Dario, 

as far as I know, you also have a less exotic scanner on site. Could you please confirm, that the behavior you describe is specific to the 9.4 T scanner? We need to understand, if this is indeed model-specific or whether we are dealing with something else. 

Also, what is really inverted, how do you observe the problem in the first place? 

--Maxim

dabosch, 2023-09-11:
Hi Maxim,

I did a few tests comparing our Prisma FIT scanner and the Magnetom 9.4T. So my findings are:

### Image orientation
for both scanners, correct and consistent image orientation can be achieved with the same .seq file using the original pulseq interpreter code.

**My solution and explanation:**
in the example `writeGradientEcho_label.m` the phase encoding gradients are counted from -k to +k in order to get the right image orientation. In other words: the PE gradient moments need to be inverted.
When expanding that example to 3D, however, it is not enough to invert the phase encoding gradients. Ice3D seems to flip the frequency encoding axis as well. So all I needed to do was invert *all* gradients in my 3D GRE, including the read gradient. 

### FOV shift
> Also, what is really inverted, how do you observe the problem in the first place?

This only affects the 9.4T, not the Prisma.

Recording an image in the isocenter works without issues on both scanners. When I shift the FOV out of isocenter in syngo the effect is applied reversed on the 9.4T. So moving the slice / FOV up will result in an image of a *lower* area. Moving it forward results in an image from more to the back, etc. I verified this with an asymmetrical phantom and a large enough FOV to observe these effects without folding.
This can be solved with the aforementioned modification to `ARBITRARY_SBB::SetFreqCoefs`.

**Possible Explanation:**
Our system actually generates 240MHz signals and mixes them with 640MHz to achieve the required 400MHz for transmit. For Rx the received 400MHz signal is again mixed with 640MHz to generate a 240MHz signal that can be sampled. So maybe we have some inverted phase sign coming from that process?

m-a-x-i-m-z, 2023-09-29:
Dear Dario, 

very interesting observations! Did you also check what is happening with the slice-selective excitations? Do they also get shifted in the opposite direction on 9.4T? 

I also wonder, whether this is common with other MR systems that use frequency mixing architecture, like 7T. Could somebody test it on 7T Magnetom Classic and also on Terra and report here? It would be really cool! 

As for flipping the readout direction or the data, you can also use the the REV label which will force ICE to flip the data in memory.  

Best regards,
Maxim