pulseq/pulseq #34: [Siemens raw data] Wrong encoding space dimension / i3DFTLength value

schuenke, 2023-09-27:
I realized that in all my siemens raw data files from 2D pulseq acquisitions, the parameter `i3DFTLength` ist set to `2` (instead of `1`). I checked this for different combinations of Pulseq versions 1.4.0 and 1.4.1 and Siemens IDEA versions VB17, VE11, XA20 and XA61.

`i3DFTLength` is used for example by the [siemens to ismrmrd converter](https://github.com/ismrmrd/siemens_to_ismrmrd) to define the [encoding matrix size](https://github.com/ismrmrd/siemens_to_ismrmrd/blob/545ee4dfb39db491f4d19e5b843225d08f216ae4/parameter_maps/IsmrmrdParameterMap_Siemens_NX.xsl#L394).

Is this a Siemens default value that cannot be changed or is there an option to define it correctly when using Pulseq and the Siemens raw data export?

QingpingChen1201, 2023-09-27:
Dear Patrick,
Many thanks for your bug report! We also found this problem in our VE11 system. For example, after the Siemens_to_Ismrmrd conversion of 2D slice-selective EPI acquisitions, the encoded matrix size of Siemens data is 96*96*1, while that of Pulseq data is 96*96*2. The EPI images reconstructed by Gadgetron contain two identical image sets. We tried to set the value of i3DFTLength to 1, but the Siemens scanner did not accept it for some reason. We will let you know once we have a fix to this ready.
Best regards,
Qingping

btasdelen, 2023-09-27:
Hi @QingpingChen1201 and @schuenke. If it helps, I was using a workaround with Gadgetron (and potentially Fire), until a proper solution is found. I wrote a small Python script that modifies the acquisition headers and passes them through down to the recon pipeline. Relevant parts are:

```Python
   for acquisition in connection:
        dshape = list(acquisition.bits[0].data.data.shape)
        dshape[2] = 1
        acquisition.bits[0].data.data = acquisition.bits[0].data.data[:,:,-1,:,:,:,:]
        acquisition.bits[0].data.data = np.reshape(acquisition.bits[0].data.data, dshape)
        acquisition.bits[0].data.sampling.encoded_matrix[2] = 1
        acquisition.bits[0].data.sampling.recon_matrix = acquisition.bits[0].data.sampling.encoded_matrix
        acquisition.bits[0].data.sampling.sampling_limits[2].max = 0
        acquisition.bits[0].data.sampling.sampling_limits[2].min = 0
        acquisition.bits[0].data.sampling.sampling_limits[2].center = 0

```
It basically sets the third dimension to 0. Unfortunately, we can not set the MRD XML headers (encoding headers to be specific), like this, because they are read-only.

QingpingChen1201, 2023-09-28:
Hallo Bilal,
Many thanks for your insightful comments. Manually modifying the value of encoded matrix size indeed solved some recon problems in Gadgetron. For Gadgetron offline recon, we can modify the MRD XML headers, like this (in Matlab):
% read the dataset
h5_file_name = 'data.h5';
dataset = ismrmrd.Dataset(h5_file_name,'dataset');
% read the header
hdr = ismrmrd.xml.deserialize(dataset.readxml);
% change encoded matrix size
hdr.encoding.encodedSpace.matrixSize.z = 1 ;
% write the header
xmlstring = ismrmrd.xml.serialize(hdr);
dataset.writexml(xmlstring);
% Write and save the dataset
dataset.close();

Hope this helps a bit:)
Cheers,
Qingping