pulseq/pulseq #35: Bug in RF time shapes

cncastillo, 2023-09-29:
Hi, I am using the XA50 version of the Pulseq, and I think I found a bug in interpreting RF waveforms when defining a time_shape waveform.

This is a minimal working example (MWE):
```matlab
% System definition
sys = mr.opts('MaxGrad', 18, 'gradUnit', 'mT/m',...
              'MaxSlew', 30, 'SlewUnit', 'T/m/s', ...
              'rfRingdownTime', 20e-6, ...
              'rfDeadTime', 100e-6, ...
              'adcDeadTime', 10e-6);
% Sequence generation
seq = mr.Sequence();
block1 = mr.makeBlockPulse(deg2rad(90),  'Duration', 1e-3/2,  'system', sys);
block2 = mr.makeBlockPulse(deg2rad(180), 'Duration', 1e-3,    'system', sys);
seq.addBlock(block1);
seq.addBlock(block2);
% Export
export_path = 'C:\ProgramData\Siemens\Numaris\MriCustomer\CustomerSeq\pulseq\';
seq.write([export_path 'test_rf.seq']);
% Plot
seq.read( [export_path 'test_rf.seq']); % To account for time discretization
seq.plot()
```
Result of plotting the function in MATLAB:
[image]

But when I simulate the sequence using POET I see this:
[image]

My intuition is that the uniqueness of each RF is probably determined based on the `mag_id` and `phase_id` but not the `time_shape_id`. I tested this by modifying the .seq file by hand adding a redundant shape, like so:
```diff
...
# Format of RF events:
# id amplitude mag_id phase_id time_shape_id delay freq phase
# ..        Hz   ....     ....          ....    us   Hz   rad
[RF]
1          500 1 2 3 100 0 0
- 2          500 1 2 4 100 0 0
+ 2          500 5 2 4 100 0 0

# Sequence Shapes
[SHAPES]

shape_id 1
num_samples 2
1
1

shape_id 2
num_samples 2
0
0

shape_id 3
num_samples 2
0
500

shape_id 4
num_samples 2
0
1000

+ shape_id 5
+ num_samples 2
+ 1
+ 1
...
```
[image]

Not sure if you have any suggestions on where to change how the RFs are read in the interpreter to account for this case :).

Cheers!

m-a-x-i-m-z, 2023-09-29:
Dear Carlos, 

thanks for such a detailed analysis and the precise diagnosis! You are absolutely right and actually I am aware of this bug exactly since one week as it was also discovered by our collaborators from Boston and reported last Friday. My bad, I should have posted it here as a known bug as this is indeed quite annoying and may have unpleasant consequences... 

The recommended WORKAROUND is now to use shaped (arbitrary) pulses with the shape containing a vector of 1 of the length corresponding to the pulse duration in place of block pulses. This will work fine for now. 

We already have a fix in our lab but unfortunately we did not branch it off from the other features of the current internal development version, where we have a bunch of other (more minor) bug-fixes and new features. We should see when we can make the next minor release that would fix the current set of known bugs...

Best regards, 
Maxim

cncastillo, 2023-10-02:
Thanks Maxim, that workaround should suffice for the time being. 

In case someone else wants to see a potential implementation of the mentioned **workaround**:

```matlab
block_dur = 1e-3; %1ms
N_block_samples = fix(block_dur / sys.rfRasterTime); 
block_samples = ones(N_block_samples, 1);
block = mr.makeArbitraryRf(block_samples, deg2rad(180), 'system', sys);
```

Cheers!

m-a-x-i-m-z, 2024-04-23:
This bug should be fixed in the current interpreter. Can someone check and confirm please?
--M.

cncastillo, 2024-04-23:
Hi Maxim, I am happy to test this. Is the newest interpreter the one in https://github.com/pulseq/pulseq/tree/master/src or should I request it via the C2P platform?

cncastillo, 2024-08-14:
Hi Maxim, sorry for the late reply. The new interpreter v1.4.3 does, in fact, solve this problem:

[image]

I will be closing this issue :)
Cheers!

closed 2024-08-14 by cncastillo