pulseq/pulseq #37: TTL trigger on GE

leRequinRomain, 2023-10-31:
I need to trigger a TTL pulse on a GE scanner. The maketrigger() function shows only physio1 and physio2 are enabled and are Siemens-specific. Is it possible to generate a trigger on GE scanners with the maketrigger() function? If not, is there an alternative method?

jfnielsen, 2023-11-01:
You can use either makeTrigger() or makeDigitalOutputPulse(), e.g., the following will produce the same output trigger signal on the GE scanner:

trig_out = mr.makeDigitalOutputPulse('ext1', 'delay', 100e-6);
trig_out = mr.makeTrigger('physio1', 'delay', 100e-6);

The 'ext1' and 'physio1' directives are simply ignored by the GE interpreter (more specifically, during conversion to the .tar file).

Note that the duration of the TTL output pulse is set by the ‘trigdur’ CV on the console, so you can vary that freely (it defaults to 500 us). You can also change the output channel on the UI.

NB! For GE (at least at the moment), a trigger cannot be the only event in a block. And the trigger start and end times can’t overlap with other instruction pulses such as those related to RF pulses or DAQ windows -- if they do you’ll get a timing error (in the simulator or on the scanner).

leRequinRomain, 2023-11-01:
Thanks for the clarification for triggering the GE system. 

I am attempting to write a radial acoustic radiation force impulse imaging (ARFI) GRE sequence for use with focused ultrasound. The Pulseq radial GRE sequence was adapted for MR-ARFI. Shown below is a snippet of the main segment that loops over each spoke and produces the following sequence diagram.

[image]

However, there are two more issues that need to be addressed:

1) First, the trigger needs to span two blocks. I have two 4 ms gradients that occur sequentially. The trigger duration should be 6 ms and span across both gradients. For example, the trigger needs to start 2 ms into the first block and remain active until the end of the second block. Can I delay the trigger event within the same block and set the 6 ms duration in the console?  For instance, seq.addBlock(gmeg1{acnt}, mr.makeDelay(2e-3),trig).

[image]

2) Second, there is an unresolved error related to the segment ID. The sequence diagram appears to be as expected, but when converting the .seq file to a .tar file for scanner upload, there is an error about the segment labels being incorrect. The segment labels are ignored for a temporary fix, but I would like to resolve the problem before testing on the scanner.

If the segment ID is set to the spoke number, the following error occurs: "1.	“Index exceeds the number of array elements. Index must not exceed 13.” 

If a constant segment ID of 2 is used, the following error is returned. "Expected parent block ID 7, found 4 (block 6)”.

jfnielsen, 2023-11-02:
Thanks for these details, this is an interesting use case and a good stress test of the GE interpreter!

1.  seq.addBlock(gmeg1{acnt}, mr.makeDelay(2e-3),trig) won't have the desired effect. The block duration is determined by the longest event, and the delay event will only serve to potentially influence the block duration. Instead, create the trig event such that it has the desired delay:

   >> trig = mr.makeDigitalOutputPulse('ext1', 'delay', 2e-3);

and then add it to the block:

  >> seq.addBlock(gmeg1{acnt}, trig)

And then, yes, I think it should be fine to have the TTL pulse extend into the next block, as long as the end of the TTL pulse doesn’t get too close to an RF/ADC event. This is possible on GE since the block boundaries ‘disappear’ inside a segment.

But the statement

  >> seq.addBlock(trig)

is not allowed currently for Pulseq on GE -- a trig event can’t be the only event inside a block.

2. I just pushed a change to the main branch of the PulCeq repo, and seq2ge.m now only warns if it detects any segment inconsistencies. This is a subtle point, but basically the previous (until today) version insisted on the waveform shapes within a segment being fixed. But I don’t think that’s strictly necessary and the interpreter shouldn’t care if, e.g., the waveform rotates. So give it a try! And happy to help, feel free to contact me.

Also, I think I was slightly wrong in my first reply. To create a TTL pulse as you described, use makeDigitalOutputPulse (makeTrigger is meant to trigger/gate the sequence via, e.g., ECG).