pulseq/pulseq #38: musicSoundTest.m example from 'demoUnsorted' folder under the 'matlab' folder will not run due to improper syntax on line 29

amoody-gh, 2023-11-16:
Line 29 uses the function musicToSequence() to convert mrMusic object to a pulse sequence. In the name-value arguments for this function, 'barDuractionSeconds' is not entered before its value (1), and 'pulseqUseWave' is not entered before its value (pulseqUseWave=false; on line 27). Fixed it using the code below: 

Bugged Code (creates runtime error):
seq = mrMusic.musicToSequence(seq, pitches, durations, 1, pulseqUseWave);

Fixed Code:
seq = mrMusic.musicToSequence(seq, pitches, durations, 'barDurationSeconds', 1, 'pulseqUseWave', pulseqUseWave);

m-a-x-i-m-z, 2023-11-16:
Good spot, thank you! As you might have guessed, we first didn't have value-pair arguments in the early implementation, and later on, after switching to the new syntax, nobody cared about the harmonic scales anymore ;-)

m-a-x-i-m-z, 2023-11-16:
just committed a fix. done.

closed 2023-11-16 by m-a-x-i-m-z

amoody-gh, 2023-11-16:
Happy to help! I’m the matlab guru at my institution so I’m used to debugging and finding errors. 

 

I look forward to your lectures tomorrow. 

 

Thank you,

Alexander Moody, MS, PhD