pulseq/pulseq #43: FA of Adiabatic RF in mprage

zhengliuer, 2024-03-11:
Hi, I'm reading writeMPRAGE.m code in seqDemo. 
https://github.com/pulseq/pulseq/blob/master/matlab/demoSeq/writeMPRAGE.m#L39 
This line tries to create a 180 flip angle adiabatic RF pulseq, I tried to calculate the flip angle of it, and I got:
```
abs(sum(rf180.signal)* (rf180.t(2)-rf180.t(1)))*360
ans =

          287.144751708818

```
Anyting I missed here? Or adiabatic RF is a little speicial and not suitable for this equation?

Thanks a lot

QingpingChen1201, 2024-03-11:
Dear Zheng,
Many thanks for your message! Adiabatic pulses are a class of amplitude- and frequency-modulated RF pulses that are relatively insensitive to B1 inhomogeneity and frequency offset effects. The design of adiabatic pulses is rather complex and utilizes the adiabatic principle. Adiabatic pulses behave differently than non-adiabatic pulses.  For example, the flip angle produced is not simply proportional to B1 magnitude and pulse length, but depends on how the B1 field varies in amplitude and phase throughout the pulse. Likewise, adiabatic pulses cannot be simply scaled or stretched to change their effect; doubling a 90° adiabatic pulse does not produce a 180° inversion pulse. Further reading: https://mriquestions.com/adiabatic-pulses.html#:~:text=Adiabatic%20pulses%20are%20a%20class,the%20B1%20field%20through%20resonance.
Therefore, the flip angle of the adiabatic RF pulse should not be calculated using the formula you mentioned. Pulseq calculates the adiabatic inversion pulse using the sigpy python package. for more details, you can have a look here: https://github.com/mikgroup/sigpy/tree/main.
Best regards,
Qingping

zhengliuer, 2024-03-11:
Thanks a lot! I will try to dig into the links.

m-a-x-i-m-z, 2024-03-11:
This is a generally interesting point to discuss (and actually not an issue), so I'm moving this to the Discussions section on GitHub and we will continue there.