pulseq/pulseq #47: multiple ADC's must be same number of samples - Siemens

bsuttonIL, 2024-04-23:
In a sequence with two different ADC's, it seems that the 2 ADC's have to have the same number of samples or the sequence crashes on the scanner. If we include two different ADC's and they have different number of points, the sequence will not run on the scanner. However, if we run with two different durations of ADC's (dwell time different) but same number of points, the sequence runs. Made sure that ADClengthpersegment in the Special card was set correctly. Also made sure that the segments would hit the gradient raster time of 10 us in each scenario.   Example: we did 920 us duration gradient and 1840 us duration gradient, set 2 us sampling giving 460 samples in first, 920 in second. Configured sequence to make sure that we would use ADClengthpersegment of 230 and all would work out. (230*2us = 460 us - on 10 us gradient raster time). sequence fails to run. But making dwell time 4 us for the second gradient (keeping 2 us for first gradient) and having 460 samples for each, sequence runs. It runs when I set the ADClengthpersegment to any number (tried 200, 230, 100) - it seems that this number may not be used in the execution of the sequence.  Thanks for the help. Glad to test other setups.

m-a-x-i-m-z, 2024-04-23:
Interesting... 
I've certainly used sequences with varying ADC length but for shorter ADC objects, which did not require ADC segmentation. It is entirely possible that the ADC segmentation code inside of the interpreter is not expecting to the number of samples to vary across ADCs. We can have a look at the problem, but it would however be best if you could send us a simple demo sequence per e-mail, as it would save us some time ;-) 
--MZ

bsuttonIL, 2024-04-23:
Thanks for taking a look, will send through email.

bsuttonIL, 2024-05-22:
Thanks to @m-a-x-i-m-z , we see that our initial issue was not related to this but instead was probably a missing timing check, just seems that we needed to add 'system',sys) to the end of every make command and then things ran. So, I can confirm that most of what my original comment indicated was not correct. We can run cartesian readouts of length 120 and non-cartesian readouts of 920 and if we set the entry ADC length per segment to 40 (which divides evenly into 120 and 920) all runs fine. ADC's are properly arrayed and can be read out of mapVBVD. Thanks!

m-a-x-i-m-z, 2024-05-29:
Excellent! Thanks for the update! I'm closing the issue.

closed 2024-05-29 by m-a-x-i-m-z