pulseq/pulseq #49: Integer overflow due to long scan times

nHimburg, 2024-05-15:
Hi,

when I try to start a pulseq sequence with a long scan time on our scanner (a Siemens system), an error message occurs, that the variable “RF_duration[0]” has a negative value. After some trial and error, I figured out that this variable seems to store the total sequence duration in μs. Unfortunately, the datatype of the variable seems to be a 32-bit signed integer. For this reason, the duration becomes negative for sequences longer than 2^31 μs~36 mins and prevents the execution of the scan.

I did, however, find a “workaround” for my sequence. By changing some timings, I can set the sequence duration to a value higher than 2*36=74 mins. This way, the rf duration is positive again and the sequence works if the gradient health check is additionally set to “None”. 

Do you maybe have an idea about the origin of the “RF_duration[0]” variable? And if so, can you recommend another way to deal with this overflow problem? I would be very happy for your help.

Best regards

Niklas

m-a-x-i-m-z, 2024-05-15:
Hi, 

thanks for your report. Older interpreters did have problems with longer sequences, but these issues should be fixed in the current one. Actually, I'm not sure I understand, which exact code snippet you are referring to... Could you be more specific and mention the file name and some relevant  line number? 

Best regards, 
Maxim & the Pulseq team

nHimburg, 2024-05-15:
Thank you for the fast answer. The Interpreter Version that we are using is 1.41
I actually don’t know where this problem occurs in the code. The details that I provided are only assumptions based on the error message that the scanner gives via logviewer: “RF-block duration negative. RFduration[0]=….”. I tried to find the relevant code snippet in the interpreter’s source code but was not successful so far.

m-a-x-i-m-z, 2024-05-15:
You should certainly try out the current release (1.4.2 from December 2023). We have fixed a bug related to an overflow of the total sequence duration counter, which was previously limiting the maximum sequence duration to about 37 minutes. So very probably you are facing the same problem and it is fixed in the current release. 

If you have access to TeamPlay, you can simply download the package. Otherwise please let us know per e-mail (pulseq.mr@...) and we will mail you the package.

nHimburg, 2024-05-27:
Dear Maxim,

we upgraded our Pulseq version to 1.4.2 and the sequence works fine now, even for a total duration > 37min.
Thank you very much for your help.

Best regards

Niklas

closed 2024-05-27 by nHimburg