pulseq/pulseq #53: Error read and add delay block

SWeinmueller, 2024-08-30:
Hi,
When I use the seq.read() function to read a .seq file, and the file contains a block with only a delay, adding this block to another seq object using seq.addBlock() results in the block not having a defined type.

I added the error:
[image]

and the script as txt file:

[delay.txt](https://github.com/user-attachments/files/16816285/delay.txt)

This could be fixed (workaround) by modifing setBlock() (https://github.com/pulseq/pulseq/blob/dd062922a523be4cac7f83a6bb56be9a3083e7db/matlab/%2Bmr/%40Sequence/Sequence.m#L567C13-L573C30):

```Matlab
for i = 1:length(varargin)
    event = varargin{i};
    % If event is just a number, it is always the block duration
    if isnumeric(event)
        fprintf("Just delay in event!")
        duration=max(duration,event);
        continue;
    end
    switch event.type
        case 'rf'
```

makeDelay(5) generates a struct with fields "type: 'delay'" and "delay: 5" while getBlock() generates a struct with fields "blockDuration: 5", "rf: []", "gx: []", "gy: []", "gz: []", "adc: []".

Best,
Simon

QingpingChen1201, 2024-08-31:
Dear Simon,
Many thanks for sharing your experience in using Pulseq with us! Yes, this is indeed a bug in the case that we use the b1 = seq.getBlock(...) to extract the block structure and then directly use seq.addBlock(b1) to build sequences again. The reason is that when we use seq.addBlock(mr.makeDelay(5.), rf, gx, gy, gz, ...) to build the block structure, we don't treat **_delay_** as a true event but only account it for **_blockDuration_**. However, this **_blockDuration_** inside the block structure cannot be recognized by the seq.setBlock(...) (it is only a number). Therefore, if we use seq.addBlock(b1) to add the whole block structure at the same time, the **_blockDuration_** will raise an error message. Yes, the method you proposed would be a good workaround in the case of adding the whole block structure. @m-a-x-i-m-z , please correct me if I am wrong:)
Best regards,
Qingping and the Pulseq team

m-a-x-i-m-z, 2024-09-24:
Yes, this is a correct description of the current status. Also, the proposed workaround is completely right. 

The question is, whether we should change the logic of the functions addBlock() / setBlock() to accept a double number automatically as if is was a delay? I don't see which negative consequences could it have. Ideas? 

Best, 
Maxim

m-a-x-i-m-z, 2024-09-24:
Yes, but I'm not sure I can see any current scenario that will be broken if we interpret any double number that is passed as a parameter to addBlock() as a delay? I mean, without disabling the current "delay" logic. I've thought about it several times, and I've only seen advantages without any negative side effects... However, it might be actually a good idea to interpret it differently, namely as a desired block duration. In this case addBlock() / setBlock() should throw an error if the desired duration is not realizable because some of the events are too long. Might actually be a nice feature...  

Btw, both proposed solutions will be compatible with re-creating a block based on another block. 

So the options are: 

1: do nothing, rely on the quick fix proposed by Qingping; 
2: interpret numeric argument as a delay; 
3: interpret numeric argument as a requested duration.

The more I think about the more I actually like the Option 3. 

--M.

QingpingChen1201, 2024-09-25:
Then I also top the Option 3 :)
Just a side note: I found that the multiple LABEL events extracted from the event block can not be passed to the addBlock(). Maybe we need to optimize the addBlock() code to make it compatible for adding LABEL events.
Best regards,
Qingping

m-a-x-i-m-z, 2024-09-30:
OK, we now accept a single numeric parameter in addBlock() and similar functions as the required block duration. With that we close the issue. Thanks for the good discussion!

closed 2024-09-30 by m-a-x-i-m-z