pulseq/pulseq #54: Specification Inconsistency When Parsing RF Dwell Time in ExternalSequence.cpp

zhengliuer, 2024-09-24:
Hello, I want to talk about something may need to be fixed about parsing RF dwell time in ExternalSequence.cpp:
https://github.com/pulseq/pulseq/blob/master/src/ExternalSequence.cpp#L1075-L1124

Code here checks if rf.time_shape_id is not 0, if not 0, calculate the rf dwell time, if is 0, then use old Pulseq's predefined RF raster time, as in 1us. However, in the specification, it says if time shape id is 0, then it means this rf event uses rf raster time as its dwell time, which is defined in mr.opts. So if users defined rf raster time as 0.1us, and the time shape id of rf is 0(uses rf raster time), when using C++ to get rf dwell time, we will have 1us, not 0.1us(rf raster time), which is an error.

Also, in `rfFromLibData`, the function in mr.Sequence, it uses rf raster time directly to calculate rf.shape_dur, which is as expected.

m-a-x-i-m-z, 2024-09-24:
Hi Zheng Liu,

Good point, thanks for spotting out. Please check the fix in the current 'dev' branch.

Best, 
Maxim

zhengliuer, 2024-09-25:
Thanks a lot!

Best regards,
Zheng Liu

closed 2024-09-25 by zhengliuer