pulseq/pulseq #64: Different behaviour for composite block pulses on different scanners

schuenke, 2024-10-31:
Hi everyone,

I am currently doing a Pulseq-based cardiac MR Fingerprinting (cMRF) multi-site study on different 3T Siemens scanners. On our scanner (Cima.X, XA61), where I tested the sequence during development, everything works fine and it get perfect matches between pulseq cMRF and reference methods. On a Skyra (XA30), the results look fine as well. Some first tests on Prisma Fit (?) (VE11) looked good as well. On a Lumina (XA50?!) however, the results deviate a lot.

In the cMRF sequence I use a t2prep block consisting of different block pulses with flipangles between 90° and 360°. The duration for a 180° is 1 ms and all other flip angles are realized by adjusting the duration. The minimum gap between two rf pulses is 150 µs.  If I disable the t2prep blocks, I get similar results for the Lumina and Cima.X scanner again (didn't test it on the other systems). So the t2prep block must be the problem!

This is how the t2prep block (MLEV-4) looks like (there is a grad spoiler at the end that I dont show here):

[image]

The amplitude of the block pulses (500 Hz) is smaller compared to other RF pulses (all sinc and one adiabatic inversion) in the sequence:
```[RF]
1      690.373 1 2 0 100 0 0
...
33          500 101 102 103 100 0 0
34          500 101 102 104 100 0 1.5708
35          500 101 102 103 100 0 3.14159
36          500 101 102 104 100 0 4.71239
37          500 101 102 105 100 0 0
38          500 101 106 107 100 0 0
...
50      576.846 3 4 0 300 0 0
51      615.672 3 4 0 300 0 0
52      654.499 3 4 0 300 0 0
53      693.325 3 4 0 300 0 0
```

I tried with different settings for `Timing and Flip Angles` (“strict”, “adapt both”) in the special card and different coils at the Lumina already. Same outcome.

Does anybody have an idea what could go wrong here? Could it be a bug related to block pulses? I once had problems with the correct scaling of block pulses in the pulseq-CEST hybrid sequence under VB17 a long time ago, but no idea if this could be related... 

I am thankful for all input including suggestions for further debugging...

m-a-x-i-m-z, 2024-10-31:
Hi! 

first of all, you should always use 'strict' if you care about flip angles staying correct. The old rect pulse bug should also be fixed since two releases. So I really don't know what is happening... 

did you try IDEA simulations for different scanners? Do they look correct?

schuenke, 2024-10-31:
Hi @m-a-x-i-m-z ,

thanks for the fast reply. Yeah, 'strict' is what we typically use for the MRF sequences, but I still gave it a try using the adapt option.

I didn't do simulations in IDEA yet. But it's a good idea! However, it will take me some time, because I don't have the environments for all the scanners installed yet. But I would expect that the Lumina/XA50 behaves similar to the Cima.X/XA61 and Skyra/XA30 if even the VE11c does, no?

Independent of the simulation results, I will hopefully be able to try a version with twice as long rf pulses and maybe also a version with sinc instead of rect pulses on Monday. If anyone has some other ideas, please let me know 😄

m-a-x-i-m-z, 2024-10-31:
One thing you can check are the pulse voltages on the scanner console, they have to be proportional to the amplitudes in the plots. 

--M.

schuenke, 2024-10-31:
Hi again,
I checked back with the people from Charité where the Lumina is located and it seems like the interpreter version on this system is still 1.4.0 as the interpreter was compiled in 2022. Is there a way to check the interpreter version at the scanner to confirm this @m-a-x-i-m-z ?

I also checked the changelogs, but didn't find any infos about the rect pulse bug. Do you know if it was already fixed in version 1.4.0? 

I will try to update the interpreter on Monday and will report if this helps for sure!

m-a-x-i-m-z, 2024-10-31:
yes, 1.4.0 was the one to introduce it and it was fixed either in 1.4.1 or 1.4.2 

so, 1.4.0 is certainly affected by the by the rect pulse bug. 

--M.

m-a-x-i-m-z, 2024-11-28:
@schuenke Could you confirm in the meantime, that the problem was related to the out-of-date interpreter installed on that scanner? We would like to close the issue ;-)

--M.

schuenke, 2024-11-28:
sorry for not updating you @m-a-x-i-m-z 

Yeah, we were able to solve the problem by updating the interpreter from 1.4.0 to 1.4.2. Unfortunately, I can't tell you if 1.4.1 would have fixed it as well, but probably there are not to many ppl around who still use 1.4.1...

closed 2024-11-28 by schuenke