pulseq/pulseq #82: [Siemens interpreter sequence] Setting TE makes the sequence unexecutbale

joergfelder, 2025-01-24:
Dear all,

I wanted to set TE with

```seq.set_definition('TE', TE)```. 
 
However, when I try to execute the sequence on the scanner, I get an image reconstruction system not ready and the log that TE should be in a range from 10ms to 10ms (which it isn't). When I remove the TE entry in the .seq file everything works as expected ... Any ideas?

Best regards
Jörg

dabosch, 2025-01-24:
Hi Jörg,

I've had the same issue in the past. I just stopped setting the parameter for now.

Best,
Dario

joergfelder, 2025-01-24:
Hi Dario,

true - that's what I am doing for the time being. However, since the TE of 10ms gets transferred into the DICOM header it would be nice to keep it consistent between measurement and DICOM - even if just for the sake of data integrity :-)

Best,
Jörg

QingpingChen1201, 2025-01-27:
Hi Jörg,
Many thanks for your continued interest in using Pulseq! I hope you are doing well in Juelich.
The reason for the "frozen" TE and TR is that after several iterations of sequence preparation,  the user-defined TE and TR are out of the allowed ranges of the protocol. Therefore, the protocol doesn't take the TE and TR values that you set using the seq.definition(...) function. The solution is to modify the related source code a bit to clarify the allowed TE and TR ranges for every iteration of the sequence preparation. We have updated the source codes for the latest Pulseq interpreter version (v1.5.0) that will come out soon. The interpreter source code is available via Siemens C2P procedure. If you would like to get the modified source code before the v1.5.0 release, please feel free to drop us an email via pulseq.mr@uniklinik-freiburg.de.
Best regards,
Qingping and the Pulseq team

joergfelder, 2025-01-27:
Dear Quingping,

I really appreciate your quick support. It is great to hear that it will work with Version 1.5.0. I am looking forward to the release. If you need someone to test I am happy to do so. If there are already enough testers on board - I'll happily wait for the official release.

Best regards,
Jörg

PS: Thanks, all kind of smooth here in Jülich. Hope you had a decent start into the year as well.

closed 2025-01-27 by joergfelder