pulseq/pulseq #87: [Siemens Interpreter] Problem for specific coil + sequence combination

schuenke, 2025-02-04:
Hi all,

we have some strange problems with the Siemens Pulseq Interpreter for a very specific combination of a receive coil and pulseq sequence combination: when we are using our 18ch flex coil, most of our spiral sequences are not working. It's just for this combination. The spiral sequences work on the same scanner with every other coil (spine, head, head+neck, 4ch flex) and we also tested the sequences on several other Siemens 3T scanners running under VB17, VE11, XA30, XA61... and the 18ch flex coil works with all other pulseq (cartesian, radial) and all other IDEA sequences (including a spiral sequence that pretty much matches our pulseq spiral sequences).

The scanner stops with an error for Run() of ArbitrarySBB in the first block including an ADC. 

Does anybody have any idea what could cause this problem?
Although I didn't really expected it to change something, I already tried slightly modified trajectories, used lower system limits (default is anyway just 15% of max_grad and ~50% of max_slew) and increased the ADC dead time. 

We are using the Siemens Interpreter v1.4.2 on our Siemens Cima.X @ XA61

schuenke, 2025-02-05:
An additional note:

The error for this combination is NOT 100% reproducible. In maybe ~5% of the cases it does actually work. All other sequences + 18ch flex and spiral seq + other coil work in 100% of the cases. We never experienced any similar issue there.

dabosch, 2025-02-05:
Strange. I've also had issues with ADC events not always running reproducable in the past.
Have you tried increasing rfRingdownTime? I had cases where I had to increase it, without understanding why. Might very well have been a different coil

It would be great if we had a better way of debugging those kind of errors. Yes, there was an error in a specific block, but which error?

schuenke, 2025-02-05:
Changing the ringdown time is on my debugging list:

- [x] increase ADC dead time --> no effect 
- [x] lower system limits (max grad & max slew) --> no effect
- [ ] increase RF ringdown time 
- [ ] replace 1st ADC by dummy delay to ensure ADC is the problem
- [ ] remove 1st readout gradients to test if grad + ADC combination is the problem

schuenke, 2025-02-05:
> Hi, As said above, you could try to increase the RFringdownTime, because different receiver coils might have different requirements on the RF ringdown time to avoid the RF signal damaging the receiver coil. 

I didn't know that the rf ringdown time is receiver coil specific. Sounds like the best guess to me atm.

> If you use ADC segmentation for your spiral readouts, there are two more things to check:
>     * Every ADC segment must align with the block duration raster time of 10 us. 
>     * The number of ADC samples in each segment must be divisible by 4.
 
I currently have < 1000 adc samples. So no segmentation. But thanks for the reminder 👍 

> If you have a multi-echo spiral readout, then there are some requirements on the Siemens ADC / digital filter. To properly initialize the filter, Siemens requires the distance between the ADCs to be at least a certain number of samples. for example, if it requires five samples and the ADC dwell time is 10 us, then the minimum distance between the two ADCs is 5*10 = 50 us. To debug this, you can try to increase the distance between two adjacent echoes or decrease ADC dwell time.

Currently it's single echo. But also good to know. Thanks.

> Did you try to track the error using Logviewer? there might be more information for your debugging. 

The error is the one from logviewer already. Unfortunately, it only tells me in which block the error was raised and not why.

> Good luck! Best regards, Qingping

Thanks. I will keep you updated.

schuenke, 2025-02-11:
Hey Bilal,

Thanks for your reply. I think selecting even/odd number of coil elements is not the solution. I did the same measurements at another scanner today with a 12ch flex coil. Again 3 volunteers and also the same issue. Auto coil selection was always 3 flex coil and 3 spine coil elements. For one volunteer it worked only when deselecting one spine element. So total of 5 elements. For the other two it only worked when deselecting one flex and one spine element. Will double check again tomorrow, but pretty sure these infos are correct.

m-a-x-i-m-z, 2025-07-22:
No, not really. 

However, are you sure, that this is a reproducible behavior (as opposed to random crashes in the prep() function that confuses the coil selection code of Siemens? 

--M.