pulseq/pulseq #96: There is a bug where the plotted label type is 'inc' and the plot does not start from the beginning of the seq.

xpjiang, 2025-03-01:
https://github.com/pulseq/pulseq/blob/7d249a0658e0a00a9d9574c54d17400775a80380/matlab/+mr/+aux/@SeqPlot/SeqPlot.m#L188
The current value of the label depends on its previous values when its type is inc. However, the existing implementation calculates this dependency ‌only when the current block is within the plot range.

xpjiang, 2025-03-03:
There is another approach for your reference.: may be we can calculate the label before the isValid condition.
like:

[image]

Best regards

m-a-x-i-m-z, 2025-03-14:
yes, we can do that. I didn't realize the plot function applies getBlock to all the block from the beginning, it is actually suboptimal and is something inherited from the initial revisions of the Pulseq file format... Well commit the corresponding change to the public repository in the coming days.

m-a-x-i-m-z, 2025-03-14:
should be closed in the latest commit, please check it.

xpjiang, 2025-03-16:
Yes, indeed, this is not an optimal solution. Ideally, if not all blocks are being drawn, then one shouldn't parse and calculate all the blocks. Thank you for implementing such an excellent tool.
Best regards