pulseq/pulseq #98: DWI EPI RECON

Farhanvrkochi, 2025-03-12:
i was trying to  implement recon for dwi epi the same code which is added in demo recon using 2d epi recon.
i could'nt get the image. how can i solve this

m-a-x-i-m-z, 2025-03-12:
Could you please be more specific? Answering all of the questions below would help us to understand your problem: 

* which sequence did you use as a template?
* what is your diffusion weighting scheme?
* are you attempting to reconstruct images offline in Matlab or other tools like BART or on the scanner? 
* if you are attempting to do a scanner integrated recon, which scanner you are using and under which software version? 
* what are your Pulseq interpreter settings? 

Also any further additional information might be helpful.

Farhanvrkochi, 2025-03-13:
Hi,

Thanks for asking for clarification. Here’s the detailed breakdown of my setup:

Sequence Template: I used the sequence provided in [writeEpiDiffusionRS.m](https://github.com/pulseq/pulseq/blob/master/matlab/demoSeq/writeEpiDiffusionRS.m) as my template.
Diffusion Weighting Scheme: The diffusion weighting parameters are those defined in the sequence file. 
Reconstruction Approach: I’m performing an offline reconstruction in MATLAB using the generated .dat file, employing the reconstruction code from [reconExample2DEPI.m](https://github.com/pulseq/pulseq/blob/master/matlab/demoRecon/reconExample2DEPI.m).
Scanner Integration: I’m not doing scanner-integrated reconstruction—I’m solely using MATLAB.

Despite following the workflow in the demo codes, I’m not getting the expected image. Could there be any adjustments needed in the recon code when working with diffusion-weighted EPI data? Perhaps there’s a mismatch in k-space trajectory calculation or scaling parameters that I should look into.

Any insights or suggestions to help diagnose this would be greatly appreciated!

QingpingChen1201, 2025-03-13:
Hi there,

Thanks for the detailed information! I'll acquire some data using the writeEpiDiffusionRS.m script and reconstruct it with the reconExample2DEPI.m script to investigate the issue.

Please keep in mind that our example sequences and reconstructions are quite basic—they are primarily designed for concept demonstration and educational purposes.

If you're aiming for high-quality images, we recommend developing a more advanced sequence and reconstruction based on the demoSeq and demoRecon scripts we provided. Of course, you are welcome to share your project and experience with other Pulseq users through our [HarmonizedMRI platform](https://harmonizedmri.github.io/) :-)

There is an advanced Pulseq-based diffusion MRI sequence for your reference: [click me](https://doi.org/10.1002/mrm.30062). 

Best regards,
Qingping and the Pulseq Team

Farhanvrkochi, 2025-03-14:
Hi, I was using the Magneton Sola VA 61.A. I was able to run the GRE.seq, GRE3D.seq, TSE, and TRUFI sequences as provided on the site, and I was also able to run the FID sequence from the demo sequence. However, I couldn’t run the diffusion EPI sequence or the related diffusion sequences in the demo. Additionally, I built some new sequences, but they fail when I try to run them on the scanner. Could you please guide me in resolving this issue? It’s worth noting that the sequences I built run smoothly in MATLAB during the error-check tests.

QingpingChen1201, 2025-03-14:
Hi,
I am not sure about the hardware limits of the Magnetom Sola. The demo diffusion EPI sequences (writeEpiDiffusionRS.m and writeEpiDiffusionRS_PMC.m) have quite high max. slew rate (180 T/m/s) and max. gradient amplitude (mT/m). In addition, diffusion sequences are quite PNS-intense (PNS=Peripheral Nerve Stimuliation). We suggest you lower the max. slew rate (e.g. to 100 T/m/s) and max. gradient strength (e.g. to 20 mT/s), and also maybe the b value (e.g. 500 s/mm^2) set in the script. One sequence that can run on one scanner doesn't guarantee it will run on another scanner. We need to carefully tune the sequence to satisfy the hardware limits.
Maybe you can also turn on the logviewer to track where the error comes from.
Best regards,
Qingping and the Pulseq team

m-a-x-i-m-z, 2025-03-14:
Hi! 

thanks for the update, good to know that some of the example sequences work for you, so your Pulseq interpreter installation on the scanner is functional. We however still need more information to be able to help you. 
* First of all,  it is unclear to me from your last post, whether you only refer to the diffusion-weighted sequence or whether there are more sequences, that don't function as expected. 
* Secondly, what to you mean by "failing"? do the sequences run through but you are unable to reconstruct an image? 
* If the latter is the case, what is the error message that you see? 
* If there is no error messages, then where is the problem? Are there any images produced? Do they have artifacts? 
* If the sequence doesn't start or doesn't run through, the recommended way of debugging this situation is to run the problematic sequences in the IDEA simulator. Please refer to the C2P documentation concerning the detailed instructions.

Farhanvrkochi, 2025-03-14:
Dear maximz and the Pulseq Team,

I am writing to seek guidance regarding an issue I encountered with my sequence. I performed a timing check in MATLAB using the following code:
%% Check sequence timing
[ok, error_report] = seq.checkTiming;
if ~ok
    fprintf('Timing check failed:\n');
    fprintf('%s\n', error_report{:});
else
    fprintf('Timing check passed successfully.\n');
end

The timing check passes successfully in MATLAB, and I am able to generate the output sequence (gx, gy, gz, rf, adc) image from my sequence. However, when I run the same sequence on the scanner, it fails.

Could you please help me identify the cause of this discrepancy or provide guidance on how to resolve the issue?

Thank you very much for your time and assistance.

m-a-x-i-m-z, 2025-03-14:
The best way to proceed is to try out the IDEA simulation.

QingpingChen1201, 2025-03-14:
Hi there,

Many thanks for your clarification! 😊 FID is indeed a very simple sequence, but as I mentioned earlier, diffusion sequences tend to be more picky. The most likely reasons for the sequence failing on your scanner are hardware limit violations or PNS (Peripheral Nerve Stimulation) violations. Please try adjusting the sequence parameters as suggested above to see if that resolves the issue.

On another note, we encourage and welcome all kinds of questions on this forum—whether simple or complex. However, we truly appreciate "well-formed" questions. When reaching out for help, it would be great if you could take a little extra time to make your query as clear and detailed as possible. This allows us to provide accurate and efficient answers while also acknowledging the time and effort we dedicate to the development and maintenance of Pulseq.

Best regards,
Qingping and the Pulseq team

QingpingChen1201, 2025-03-14:
regarding "also is it able to simulate seq file from matlab in idea ?", please check the C2P manual for the detailed instructions about sequence simulation.
Best regards,
Qingping and the Pulseq team

closed 2025-03-14 by Farhanvrkochi