pulseq/pulseq #104: [Siemens Interpreter] Pulseq v1.5.0 not compatible with older seq files containing labels

schuenke, 2025-04-01:
Good morning,

we just installed Pulseq 1.5.0 on our Cima.X and I am facing the following problem:

When I want to run sequences containing labels, the scanner aborts the measurement with the following error:
`fSeqRun: *** ERROR: Label/counter is out of bounds for block X. Message source .../pulseqv150/Arbitrary.cpp:2656`

This happens for ALL sequences containing labels, including the `gre_example.seq` included in the source code.
Switching the `Use SET for SLC` option in the Special Card does not help. 

When running the [PyPulseq gre_example without labels](https://github.com/imr-framework/pypulseq/blob/master/examples/scripts/write_gre.py) or the `fid_builtin.seq` or using the `built-in` option, everything works. 

Is this a known issue? I would assume the new version is supposed to be backwards compatible, no?

m-a-x-i-m-z, 2025-04-01:
Hi!

Hm, we used to have a bug for a while that counters/labels that were not explicitly initialized to 0 at the start of the sequence would produce errors similar to what you report. I've thought that we've fixed it, but maybe it is not the case. 

Could you please try to set all counters to 0 explicitly at the start? 

Best regards, 
Maxim

m-a-x-i-m-z, 2025-04-01:
A more proper fix is to comment out the line 1526 (or around, depending on the exact revision of your code) in Arbitrary.cpp, which should read then: 

/* if (arbitrarySBB.getLabelStorage().adcLabelsInUse()) */ {
...

Meaning that the next line (not shown) should run unconditionally.

--M.

schuenke, 2025-07-22:
> A more proper fix is to comment out the line 1526 (or around, depending on the exact revision of your code) in Arbitrary.cpp, which should read then:
> 
> /* if (arbitrarySBB.getLabelStorage().adcLabelsInUse()) */ { ...
> 
> Meaning that the next line (not shown) should run unconditionally.

> Hi, I can confirm that the GRE example provided with pulseq runs now on our Cima.X applying [the patch Maxim suggested](https://github.com/pulseq/pulseq/issues/104#issuecomment-2769385592).

With the fix from Maxim, everything seems to work fine now. Thanks again.

closed 2025-07-22 by schuenke