pulseq/pulseq #115: Inaccurate phase plots for block pulses

schuenke, 2025-05-23:
Currently, for block / rectangular pulses, only the 2 edge points are considered in the plot function. Thus, if a block pulse has a frequency offset, only the start and end phase values will be calculated and connected with a straight line in the sequence plot. This is demonstrated in this mini example including a block and a sinc pulse with identical settings:

[image]

Adding an interpolation step like this for block pulses with offresonance != 0  can "fix" this:

```Matlab                        
% If off-resonant and rectangular (2 samples), interpolate the pulse
if (length(s) == 2) && (abs(rf.freqOffset) > 0 || abs(rf.freqPPM) > 0)
    numInterp = min(ceil(rf.freqOffset), 256);
    t = linspace(rf.t(1), rf.t(end), numInterp).';
    s = s(1) * ones(numInterp, 1);
end
```
Some comments:

- we dont have the rf.freqPPM attribute in Python yet. So I am not really familiar with it and more or less ignored it until now
- the number of interpolation steps can ofc be adapted. Using a maximum of 256 values, but less for "slow" modulations (due to low freq offsets) seemed to be a good idea to me. 

The output after adding the code looks like this:
 
[image]

And for completeness, here is the code for my mini example:

```Matlab                        
system = mr.opts();
seq=mr.Sequence(system);

freqOffset = 10 * system.gamma * 1e-6 * 3;
rf_rect = mr.makeBlockPulse(pi,'Duration',4e-3, 'system', system,'use','excitation', 'freqOffset', freqOffset);
rf_sinc = mr.makeSincPulse(pi,'Duration',4e-3, 'system', system,'use','excitation', 'freqOffset', freqOffset);

seq.addBlock(rf_rect);
seq.addBlock(mr.makeDelay(1e-4))
seq.addBlock(rf_sinc)

seq.plot()
end
```

I wanted to discuss this first, but could create the PRs for Matlab and Python if we agree on a good solution 😄

m-a-x-i-m-z, 2025-05-23:
Hi Partick, 

good spot, congratulations! 

As for the solution: I think we  need to add intermediate samples for off-resonance pulses. 

Best regards,
Maxim

mzaiss, 2025-05-23:
I think this was working in 1.3. and must have been lost when explicit /flexible sampling was introduced in 1.4.

QingpingChen1201, 2025-05-28:
Hi,
Many thanks for pointing this out. I have updated the `mr.aux.Seqplot` function: https://github.com/pulseq/pulseq/blob/master/matlab/%2Bmr/%2Baux/%40SeqPlot/SeqPlot.m
to fix this inaccurate phase plot for RF pulses with two points by adding interpolation.
The modification in our internal git is below:
[image]
Please feel free to let us know if you have any further suggestions.
Best regards,
Qingping and the Pulseq team

QingpingChen1201, 2025-05-28:
Many thanks for the suggestions! I have updated our internal git accordingly, as below:
[image]
Best regards,
Qingping

mzaiss, 2025-06-06:
> do phase plots work as intended now?

The work around with some samples looks good. but maybe the sampling for
the plot should be frequency dependent. So very large freq offsets, e.g.
bloch siegert etc don't get wrapped wrongly.