pulseq/pulseq #142: Interference between two ADC events defined within a TR

namgyunlee, 2025-07-31:
Dear Pulseq team,

I have implemented bSSFP sequences with and without self-navigation. The ADC event for self-navigation uses a dwell time of 10 usec, which is different from the dwell time of an imaging ADC event.

[image]
[image]

When self-navigation is on, images contain a line artifact at the center of a stack, which seems to be caused by a mixing of two ADC data.

[image]
[image]

I am using the Siemens interpreter (v1.5.0) at 0.55T Aera.

I have observed the same behavior in Pulseq files created with and without LABEL extension.

I have tried to label self-navigation as either "NAV" or the second echo using "ECO" LABEL. But all images contained the same artifact.
 
I would like to understand the root cause of this because FID navigators would be very useful for many different applications.

Thank you so much.

Sincerely,
Nam Lee

m-a-x-i-m-z, 2025-07-31:
Interesting. Do I understand you right, you add navigator ADC while keeping the entire timing identical and then you observe the artifact in the main imaging data. is that right?

So, how the navigator data look like? Are they reasonable/usable? 

The reason why I'm asking is that we have observed artifacts in the ADC data when the ADC object was too close to the following RF pulse.

Another idea I have is related to the initialization of digital filters. We have observed that sometimes it helps to acquire more samples with denser dwell times. So, try increasing oversampling, i.e. keeping the duration of the ADC(s) the same but increasing the number of samples. You can try doing this with both main ADC and the FID navigator.  I actually also think, it would be advantageous for the digital filters inside of the ADC chips to make sure that both ADCs have the same dwell time. 

Finally, you may try keeping everything the same in terms of the data but adding some delays around the ADCs to shift them away from the RF pulses or from each other. 

--M.

namgyunlee, 2025-07-31:
Dear Maxim,

Thank you so much for your quick response.

Do I understand you right, you add navigator ADC while keeping the entire timing identical and then you observe the artifact in the main imaging data. is that right?
=> Yes, that is correct. I have tried to place the navigator ADC event at two different locations: 1) after a slice-rephasing gradient lobe, and 2) after phase-encoding/slice-encoding rewinders. Whenever the navigator ADC event was added, I observed the artifact in the main imaging data.

So, how the navigator data look like? Are they reasonable/usable?
=> So far I have tested this sequence only on the phantom. When a navigator ADC event was placed right after the slice-rephasing gradient lobe, FID navigator data looks fine since the signal fluctuation is correlated with the pattern of phase encoding steps. I have acquired only 12 sample points for the FID navigator (discard_pre/actual sample/discard_post = 4/4/4). Please see attached.

[image]

I will implement your recommendations (oversampling, same dwell time, adding delays around the ADCs) and keep you posted.

--Nam

namgyunlee, 2025-08-05:
Dear Maxim,

I have tried all of your recommendations: 1) 512 samples for the FID navigator, 2) identical dwell time, and 3) adding delays around ADCs. With your recommendations, the line artifact is reduced. However, there are still remaining, reduced line artifacts both in the individual coil images and coil-combined images (Please see attached).

[image]

[image]

[image]

[image]

[image]

[image]

Since my test cases are in essence similar to multi-echo bSSFP imaging, does it mean any multi-echo sequences would be suffered from this line artifact? 

Thank you so much.

Sincerely,
Nam Lee